ardyesp/ESPWebDAV #2: SD card speed

set4up, 2018-12-19:
Hello. Can I somehow increase the speed of work with the SD card? Now I have a speed of approximately 176kb / s.

ardyesp, 2019-01-26:
The speed is limited by amount of RAM available in ESP8266. You can try port it over to ESP32 to get better results.

set4up, 2019-01-27:
The speed now worries me less, I had another problem: over time, not all of the file was first written, and then it completely stopped writing.

ardyesp, 2019-01-27:
It’s been working flawlessly for me. I would say check the wiring for loose connection or maybe a new SD card. 
You can check SDFat library examples and try to read/write directly, without WiFi connection.

set4up, 2019-01-27:
Good idea. thank

>Воскресенье, 27 января 2019, 18:19 +03:00 от ardyesp <notifications@github.com>:
>
>Это работает безупречно для меня.  Я бы сказал, проверьте проводку на предмет слабого соединения или, может быть, новую SD-карту. 
>Вы можете проверить примеры библиотек SDFat и попробовать читать / писать напрямую, без подключения к WiFi. 
>
>
>> 27 января 2019 года, в 3:12, set4up <  notifications@github.com > wrote: 
>>> 
>Скорость теперь беспокоит меня меньше, у меня возникла другая проблема: со временем не весь файл был впервые написан, и тогда это полностью прекратило писать. 
>>> 
>- 
>> Вы получаете это, потому что вы прокомментировали. 
>> Ответьте на это письмо напрямую, просмотрите его на GitHub или отключите тему.
>- 
>Вы получаете это, потому что вы автор темы. 
>Ответьте на это письмо напрямую,  просмотрите его на GitHub или  отключите тему .

С уважением, Евгений

closed 2019-09-15 by ardyesp