ardyesp/ESPWebDAV #11: compiler problems

pvdw0310, 2020-12-11:
Hi, 
I like to compile printer-3d.ino and i have different compiler errors :  I added the library (sdfat) form github, but the compiler doesn't  find  library, like the sdfat.h . 

what did I wrong?
thanks

luc-github, 2021-01-15:
you cannot have SdFat-master as installed library - it conflict with the embedded version of SDFat in esp8266 - remove the sdfat-master library and it will remove conflict

pvdw0310, 2021-01-16:
Thanks for the reply, but now I have still a compiler error.

Arduino: 1.8.13 (Windows 7), Board:"WeMos D1 R1, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

```In file included from C:\Users\peter\Documents\Arduino\Printer_3D\Printer_3D.ino:5:0:
C:\Users\peter\Documents\Arduino\libraries\ESPWebDAV-master/ESPWebDAV.h:39:43: error: 'FatFile' has not been declared
  void sendPropResponse(boolean recursing, FatFile *curFile);
                                           ^
C:\Users\peter\Documents\Arduino\libraries\ESPWebDAV-master/ESPWebDAV.h:42:40: error: 'FatFile' has not been declared
  void handleWriteError(String message, FatFile *wFile);
                                        ^
C:\Users\peter\Documents\Arduino\libraries\ESPWebDAV-master/ESPWebDAV.h:64:2: error: 'SdFat' does not name a type
  SdFat sd;
  ^
Bibliotheek ESP8266WiFi op versie 1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\ESP8266WiFi  wordt gebruikt
Bibliotheek ESPWebDAV-master in map: C:\Users\peter\Documents\Arduino\libraries\ESPWebDAV-master (legacy) wordt gebruikt
Bibliotheek ESP8266SdFat op versie 1.1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\ESP8266SdFat  wordt gebruikt
Bibliotheek SPI op versie 1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\SPI  wordt gebruikt
Bibliotheek Hash op versie 1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\Hash  wordt gebruikt
exit status 1
Fout bij het compileren voor board WeMos D1 R1
```

acguardia, 2021-05-09:
Hello.
Use esp8266 board version 2.4.0 -> https://github.com/esp8266/Arduino
[image]

Use SdFat library version 1.0.0 -> https://github.com/greiman/SdFat
[image]