ardyesp/ESPWebDAV #16: Won't work in non merlin systems like a digital photo frame.

devicemodder, 2021-05-14:
I bought this card from aliexpress to use in a digital picture frame, so i can send pictures to it over wifi. Problem is, it doesn't always show the SD card in the network share when it's in the picture frame as the picture frame or any device with an SD slot is always trying to access the SD card. is there a way that the firmware can be modified to work for my application? I bought it thinking that it would work like a regular NAS, allowing me to copy files to it over wifi while the card is mounted in another device.

ardyesp, 2021-05-24:
No, if the device using the SDCard never relinquishes the control of the card, then this module will never get a chance to access it. You could try to modify the firmware to write to card, despite the picture frame accessing it, but it might lead to file contention and write errors on the card. 
I would suggest that you look for an option on the picture frame to let it give up card control in between.

closed 2021-05-24 by ardyesp

devicemodder, 2021-05-24:
Thanks, that's what i figured. I'll, see if i can find an option on the frame or some way to get it working.